kendallgoto/switchbota #1: Can I download a new flash upgrade through OTA?

linagee, 2022-05-08:
I got this to work perfectly with my SwitchBot Plug Mini Plug (W1901400). But now that I have Tasmota, can I upgrade? I get "Program flash size is larger than real flash size" when I try to use the Tasmota webUI to upgrade firmwares. I understand this is because the device has 2MB and 4MB is really kind of required for OTA to be a nice process. (I understand that your app.bin is a workaround for this.)

I tried using Tasmota to load app.bin again in the hopes that it will retrieve my http://www.wohand.com/payload.bin again. I got the same "flash size is larger than real flash size". (Even though your app.bin is 800kB (half of 2MB), I guess it still doesn't like that?)

Do you know of a way to make this work? Will I need an even more minimal version than your app.bin? Or is Tasmota just being really really safe saying "No!" and maybe there's a way to bypass that?

kendallgoto, 2022-05-08:
I haven't played with upgrading Tasmota over OTA & am not particularly familiar with Tasmota's actual OTA process. Sadly I'm traveling currently and can't test it hands-on at the moment.

However: the stock Tasmota version being loaded is the special 2MB build of Tasmota32c3 in the unofficial binary list (https://github.com/tasmota/install/blob/main/firmware/unofficial/tasmota32c3_2M.factory.bin). Usually for limited space OTAs, you need to downgrade from tasmota regular to tasmota-minimal. I don't believe a "minimal" version exists for the 32c3-2M build.

I'm not sure where the build configuration used for the 32c3_2M build lies & how "minimal" it already is. It's possible that a custom Tasmota build can be made to OTA to the latest 32c3_2M version. Else, you'd likely need to flash a minimal app image that just performs the OTA (similar to the Espressif code in this repo, except rebuilt for the Arduino ESP framework instead of Espressif's ESP-IDF firmware.

I think you're confused about the purposes of the app.bin from this repo vs. Tasmota binaries -- the app.bin here no longer will work at all once the plug has been flashed to Tasmota. The devices come from factory running the Espressif bootloader + core (ESP-IDF) whereas Tasmota runs on the Arduino framework. The app.bin is an ESP-IDF app image that flashes the plug into the Arduino framework by installing a Tasmota full image. Once you're in the Tasmota image / arduino framework, the `app.bin` will never properly flash to the plug (and if it did, the plug wouldn't boot up again). To perform an OTA, you'd have to rely fully on Tasmota binaries -- whether that's existing builds or a custom "super-minimal" build. I'd love to play around and look further into this, but my plugs have been working excellently on the existing Tasmota version so I haven't had the time to play around.

linagee, 2022-05-08:
Delving into this probably way too deep, but:
tasmota/xdrv_01_webserver.ino:2781
```
        if (0xE9 == upload.buf[0]) {
          uint32_t bin_flash_size = ESP.magicFlashChipSize((upload.buf[3] & 0xf0) >> 4);
          if (bin_flash_size > ESP.getFlashChipRealSize()) {
            Web.upload_error = 4;  // Program flash size is larger than real flash size
            return;
          }
```
Reads the first four bytes of the file and gets the ESP File Information Header. app.bin says "E9 06 02 2F". The 2 of the 2F says this is a file for a 4MB flash. (And, physically, we have a 2MB flash chip.) (Source: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Esp.cpp#L335 ) So I guess that's the reason for the "Program flash size is larger than real flash size" error.
I tried changing my app.bin to "E9 06 02 1F". Surprisingly, I was able to get past this error, but now it says "Not enough space".

(Which means I have angered the Update.begin() or Update.write() functions. Both are part of the standard ESP32 library, but it's a mess to see how these are checked/computed.) I stopped here, it's probably not going any further easily.

> & am not particularly familiar with Tasmota's actual OTA process

I agree with you 100% here! :+1: 

> Sadly I'm traveling currently and can't test it hands-on at the moment.

Your response was rather quick and I appreciate it.

> the stock Tasmota version being loaded is the special 2MB build of Tasmota32c3 in the unofficial binary list

Ohh.... the special 2MB Tasmota has this header: "E9 03 03 1F". I tried it once more with the special 2MB build (1.5MB size) and I still get "Not enough space". Interesting, the 4MB builds are also 1.5MB, I guess this one just has the changed header maybe?

> I don't believe a "minimal" version exists for the 32c3-2M build.

I don't think so either, but the app.bin in your project is 800kB and I think, if I can make it work, that would be a nice minimal OTA loader.

>  Once you're in the Tasmota image / arduino framework, the app.bin will never properly flash to the plug (and if it did, the plug wouldn't boot up again). 

Is this just because of the partition layout, or something different? I was hoping it would see the NVS as invalid and be able to use the switchbota/switchbota SSID.

It might be worth cracking mine open and trying manual flashing to get further with this, but I hear it's a pain to get it open and a pain to get TX/RX/GND/GPIO0/etc soldered. (I guess it has to be dremeled or cracked open with force?)

kendallgoto, 2022-05-08:
> I don't think so either, but the app.bin in your project is 800kB and I think, if I can make it work, that would be a nice minimal OTA loader.

I'm not sure how the Arduino / ESP-IDF frameworks differ. I believe the Arduino framework is a wrapper around the ESP-IDF framework, with helpers to allow for developing app code in Arduino C / .ino files versus the straight C/C++ in the IDF. I've only personally worked on the ESP-IDF space - I know very little about the Arduino version. However, the partition table and bootloader code in the Arduino / IDF frameworks differ and are completely incompatible AFAIK. That's why this project's app.bin replaces the plug's default IDF bootloader with an Arduino bootloader. If Arduino app images were compatible directly, we could directly flash Tasmota's app image easily without the intermediate firmware from `app.bin` in this repo.

> It might be worth cracking mine open and trying manual flashing to get further with this, but I hear it's a pain to get it open and a pain to get TX/RX/GND/GPIO0/etc soldered. (I guess it has to be dremeled or cracked open with force?)

It's not too hard to get them open, but it does probably take a dremel. I didn't find a good way to split any of my 50 plugs by force ... on a lot of other plugs, there's gaps that you can shove a tool into to break the seals, but SwitchBot did a pretty good job on these. I summarized my teardown on the Tasmota template repo page incl. remarks on where to cut to solder (https://templates.blakadder.com/switchbot_plugmini_W1901400.html)

I can take a look at the end of the month when I'm back home, but strictly speaking a minimal ESP32-C3 version of Tasmota / a minimal app image to perform the OTA isn't within the scope of this project. Ideally, as official support for the ESP32-C3 grows on the Tasmota side, they'll be able to provide more information on how to perform the OTA. You're welcome to raise an issue with them as well, though I don't think they're particularly interested in providing support for the 2MB flash C3 at this time.

Jason2866, 2022-05-08:
There are changes in OTA update ahead! We did not like the waste of space for OTA having both OTA slots the same size. So we use a factory partition to store a safeboot firmware (only job is doing OTA upgrade) which is small in size (<800kbyte).
For OTA Tasmota switches to this partition and loads the new firmware in the App partition (OTA1). There is in future only one! firmware (app) partition. The new partition scheme is [here](https://github.com/arendst/Tasmota/blob/development/esp32_partition_app1856k_fs1344k.csv) All development builds (with 4MB flash) do already use.
There will be soon a Migration tool to convert already running Tasmota32 devices to the new partitition scheme.

Jason2866, 2022-05-08:
Forgot, slick way to convert to Tasmota!