kendallgoto/switchbota #4: Cant get switchbota to work with switch on v1.2 firmware

triggerx, 2022-05-10:
I have 3 plugs, two were on Firmware Version 1.1 and switchbota worked perfectly on those to upgrade to Tasmota.  However, the third one is on V1.2 and doesnt seem to be doing anything with the app.bin.  I can see the web server sending app.bin, so I can see it trying, but it's just not going.  Is this currently only working with Firmware Version 1.1?  Thanks.

triggerx, 2022-05-10:
Your bluetooth instructions on the other issue worked brilliantly to resolve my issue!  3 for 3 now.  Thanks!

handonam, 2022-05-10:
For reference, in case anyone was wondering to the above:

https://github.com/kendallgoto/switchbota/issues/3#issuecomment-1121864522

closed 2022-05-10 by kendallgoto