kendallgoto/switchbota #7: Flash size

Cossid, 2022-05-19:
I noticed you are specifically calling the 2M flash.  I had one of these opened up from before you released this, and esptool reported a 4M flash.  I'm wondering if you were just making an assumption, or did you find something different?  I was able to flash a 4M Tasmota onto mine via usb-serial without issue.

Here is an esptool info dump from mine:

`esptool.py.exe flash_id
esptool.py v3.3
Found 1 serial ports
Serial port COM3
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 60:55:f9:##:##:##
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...`

kendallgoto, 2022-05-20:
that's interesting! looking back at my notes, i see that my units reported 4M as well, despite my presumption that the units were 2M flash.

Looking now, I have a theory for how this happened as a newbie to Tasmota:
I used the 2M version due to the way it shows up on the [Web Flasher](https://tasmota.github.io/install/) -- without regard to the device filter on the right, I used the only version labeled "32-C3", assuming there was only a 2MB version of Tasmota built for the C3. I know now this isn't the case, though it seems I didn't put the pieces together at the time. I guess at some point, on account of this, I forgot the original esptool output and 4MB flash size. Not entirely sure how that happened - perhaps I was confused because the OTA images are each ~2MB, whereas the flash itself is 4MB of course, since there are two OTA images.

However, for OTA, there may still be a constraint preventing a 4MB tasmota image. I am currently unavailable to test to confirm my suspicions, but from memory:
Right now we are flashing the full 2MB image on the lower half of the flash. The upper half gets completely ignored when the device next boots. We're able to flash a full 2MB factory image since our second OTA partition, where our running code lies, starts at offset 0x220000 / 2.2MB; thus we do not run into corruption issues by erasing active instructions off of the flash. However, on a 4MB "factory" binary, we would not have this luxury since Tasmota's second OTA partition would likely line up with our running partition. In order to make this viable, we'd have to flash the bootloader and first OTA partition manually, in essence a 2MB segment of the 4MB variant.

I'm a bit rusty on the subject so take this with a grain of salt. @Jason2866 - are you able to validate my understanding reg. flashing the default 4MB Tasmota partition table?

I'd love to get this code adjusted to flash the 4MB variant, but I'm not sure it's as simple as switching out the binary. I can test a solution hands-on next week & possibly determine a way to upgrade existing 2M plugs to the 4M build. This would resolve the concerns in #1, though I'm not sure what other additional tasmota functionality would be unlocked (again, i'm a novice to tasmota itself and its use cases).

Cossid, 2022-05-20:
Yeah, I've been exploring ways to switch from 2M to 4M via OTA, but have not been successful.  I've tried doing small factory esphome images, and even they won't work at under 500k, so it's not a space limitation, but a partition limitation it seems.

I fully understand that we may be somewhat limited by the partition alignment from the factory image when it comes to OTA, but it would be nice to get the full potential/future-proofing if possible.  It doesn't look like Tasmota is too close to running into OTA restrictions for 2M->2M.  I was mostly looking for the insight that led to this decision, so thanks for that.

It is weird the way the 2M version reports space though:
```
Flash Size	2048 kB
Program Size	1408 kB
Free Program Space	1792 kB
Free Memory	201.5 kB (frag. 40%)
```

Cossid, 2022-05-20:
Well, I was going to try to revert my open one back to stock firmware and see if 4M would flash from stock firmware using your tool, but I accidentally did a read_flash instead of write_flash and overwrote my only stock backup.  If you happen to know where I could get a hold of the stock I'd be willing to give it a try.  the Switchbot OTA bins don't work, they don't appear to be app-only, no bootloader or partition info.

kendallgoto, 2022-05-20:
I'll follow up next week. I have a box of new units & the original flash archived on my home system.

If you're interested in an adventure, you can also just flash the bootloader and partition table from this project, although you'd need to setup the IDF to build them, since they're the same. The app.bin for this project is only the OTA partition, much like the Switchbot binaries you had.

Cossid, 2022-05-20:
Thanks for the direction.  It took a bit of fumbling around, but I think I mostly figured it out.  I couldn't figure out how to package an ota bin into it, but I didn't get it to create a full 4M image, flashed it, then flashed the ota bin at ota_0's offset, and that appears to have gotten it back to stock configuration.

Unfortunately, attempting to flash the full C3 4M factory tasmota bin puts it in a loop where it keeps trying to pull /payload.bin so it looks like this can't be handled by stock OTA to 4M.

It looks like we'd probably need something that could flash a safeboot compilation that knows how to reset partitioning and bootloader, but that's certainly outside my realm of know-how.

Jason2866, 2022-05-20:
A few things needs to fit together to use the whole flash. 
- the Arduino Bootloader at the fixed address 0x1000
- a partition scheme which represents the actual partition config at 0x8000
- boot_app0.bin at 0xe000
- user firmware at one of the addresses listed for the OTAs in partition
- enough space in the two OTA partitions
Flashing a factory image in one of the OTA slots is a very special approach.
I am still wondering that this is working! Since flashing a factory image via OTA places all the mentioned files with the offsets i wrote at the beginning of the OTA address.
So it seems relative addresses are used.
For further investigations a full flash firmware dump of a Tasmotized device would help.
(i cant analyze the next 4 weeks, just online via mobile)

Cossid, 2022-05-20:
@Jason2866 Here is a dump from my compiled switchbota partition layout for whenever you have time.  I'm not 100% certain if this is the same as stock, so hopefully @kendallgoto can follow up later with a verified version.

[switchbot_ota_to_tasmota.bin.gz](https://github.com/kendallgoto/switchbota/files/8736917/switchbot_ota_to_tasmota.bin.gz)

Jason2866, 2022-05-20:
Mhh, looks not as i thought. The needed files are all on there usual place. Does the Tasmota filesystem work?
Normally it should. Upload the [Tasmota Partition Manager tapp](https://github.com/Jason2866/Tasmota-build/blob/special_builder/tasmota/berry/modules/Partition_Manager.tapp) If this version does not work you have to search the one which was in the repo at March 31 (thats your build time of Tasmota i see in the dump). In this case use PartitionManager.be version (You could rename to autoexec.be so it loads after a restart).
Now check if you can change the partition sizes with partition manager. If yes, do a change. Maybe just 64k. It does not really matter (dont make the filesystem to small!!!). Do the changes and restart. If this worked there is a chance that you have now the real flash size shown. A problem is that the Tasmota version is not a actual one. With changes made one or two weeks ago, we could use Partition Wizard which does check for real flash size found (as esptool) and patches the magic bytes in the firmware. But as said i cant test anything in real....

Cossid, 2022-05-20:
Tasmota filesystem is enabled.
I did indeed need to use the version of Partition_Manager.tapp from 3/31 before it would even open (no real surprise there).
It does not look like there is anything I can do from the GUI:
[image]
The "Current Active Partition" button is disabled.
Are there commands available for Partition Manager?  I wasn't able to find any documentation.

Cossid, 2022-05-21:
After a bit more research, it appears the app.bin binary used to do the OTA flash is tied to a single MD5.  As I was unsuccessful flashing 4M, I was also unsuccessful flashing a newer 2M factory binary.

I'll look further into compiling custom app.bin files and retest both newer 2M and 4M builds.

@kendallgoto It appears when you repeated download a file that doesn't match the MD5, it never makes it into the fallback switchbota/switchbota mode, it just keeps trying payload.bin continuously.

kendallgoto, 2022-05-21:
Aha! I forgot about that complication. Yes, to avoid corrupt flashing bricking a device it repeats attempts if an MD5 checksum does not match, before rebooting; as after rebooting a device may be permanently bricked if the flash fails.

Again, I'm happy to contribute on this front and build/test with the 4M binary, but not until Monday.

Cossid, 2022-05-21:
@kendallgoto no issues.  I'm getting pretty familiar with this environment now.  I have actually just tested, and OTA flashing from Switchbot to the 4M binary works just fine with the correct app.bin!  (I just bypassed the md5 checksum check for now, I know that's not the correct solution, but it works fine for the device I can physically flash).

Now I'm going to have to look into what it is going to take to get anything already flashed to 2M Tasmota converted to 4M.

kendallgoto, 2022-05-21:
neat! i'll verify and get that change made shortly with the corrected MD5 etc. 

Now that I think about it, the erasing part of the flash only erases the first 2Mb of memory, meaning it doesn't even try to erase the actively running part. By then flashing over it, we in essence create a "partial OTA" image I was previously describing - this behavior may work but not be stable per se.

Are you able to perform an OTA on the newly flashed 4M version (ie just OTA update the same Tasmota build via Tasmota. The 2M tasmota version doesn't support OTA)? This was an issue in #1 that I'd love to verify as fixed with this solution. 

converting the 2M models over might be a harder challenge. with some trickery, we might be able to leverage the unpartitioned 2MB of space to store a lightweight image to handle re-flashing the devices. I have about 40 plugs currently factory sealed and running the 2M version so I have a vested interested. Please let me know if you discover anything and I'll look closer next week.

Cossid, 2022-05-21:
> Are you able to perform an OTA on the newly flashed 4M version (ie just OTA update the same Tasmota build via Tasmota. The 2M tasmota version doesn't support OTA)?

Yes, I specifically went from SwitchBot OTA to 11.1.0 release so that I could then OTA to 11.1.0.3 dev, and all three of those steps work fine.

Yeah, converting the 2M won't be easy, if even possible, it will all likely revolve around getting a custom partition manager tapp that can work on the active partition.  I'll look into it as time allows, but I'll need to do a bit more learning first.

Jason2866, 2022-05-21:
Latest Tasmota do not use 2 OTA partitions anymore. It uses a factory firmware to update the one OTA partition.
You may try https://github.com/arendst/Tasmota/tree/development/tasmota/berry/modules/Partition_Wizard
It is a tool to convert the legacy partition scheme (2 OTAs with same size) into the new partition scheme 
- 1 partition factory firmware (small in size) to do the OTA process and failsafe firmware
- 1 OTA partition with increased size
- 1 filesystem partition size not changed, content keeps intact in migration process
There are safeguards in the Partition Wizard. It is written in Berry.
If you have a look you will understand since you are coders :-)

Cossid, 2022-05-21:
Thanks for the guidance Jason2866.  I did try loading Partition_Wizard, but unfortunately, it is too new for our case of having to execute from 11.1.0.1.  It has reference errors to partition_core when used on this build, and I haven't make it far enough through code to see if that can be overcome.  This is my first foray into Berry, so it will take some poking around to figure out what can and can't be used.  It looks like Partition_Manager does have resize capabilities, so I think I will try that first, though I'm still just in the phase of learning to execute simple things, much less large classes like this and learning what the difference is between be (code) and tapp (binary), and how to load each.  I'll keep plugging away, and reach out if I get hung up on anything.

Cossid, 2022-05-22:
Preliminary findings are not very promising.  I made adjustments to partition manager, but I was unable to make any partition modifications, anything I tried to do, including shrinking or expanding the SPIFFS partition by 16KB always resulted in

> Exception:
'internal_error'
Error calling spi_flash_read()

So I adapted partition_core.be and partition_wizard.be with some forcing of flash sizes where functions didn't exist, and anytime flash.write is hit, the same error

> Exception:
'internal_error'
Error calling spi_flash_read()

occurs.  I tried following the code as to why there is a spi_flash_read() in the flash.write() call, but I couldn't fully track it.  I'm assuming there is some verification read that happens or something, and that is failing.  I have no idea how to get past such a thing, as those appear to be ESP32 API interface extern calls.

Additionally, I tried creating an switchbota app.bin file with a bunch of compiler flags stripped out to reduce size and was able to get a gz down to 360k (there is 384k free on the app0 partition on this device), but it would fail with out of space.  I'm guessing gz requires two app partitions to extract to before executing, but I'm unsure.  I was getting uncompressed app.bin files down to about 570k, but they would fail stating the flash sizes didn't match, so tasmota has some protections built in that we probably can't get around on that front.

I'm running a bit out of ideas at the moment.

Jason2866, 2022-05-22:
gz is not supported from arduino esp32x bootloader/OTA. This can not work.
It could be that the version 31/3 has no support for writing to flash outside his own area.
(needed sdkconfig for has to be compiled in Arduino libs -> "Dangerous writes allowed")
If this is the case there is **NO** chance to change partition size without a serial flash.
The device needs a firmware running which has this feature enabled.
So a solution for OTA partition resizing is only for users who has flashed a firmware with this feature. Afair it was introduced for all boards around 10th of May.

Cossid, 2022-05-22:
Why would I get the same spi_read_flash() errors when trying to shrink app0 or the SPIFFS partitions?

Jason2866, 2022-05-22:
writing to this sections is only possible via the provided API. Example for SPIFFS you use the provided read write functions. Not directly writing to flash addresses.

linagee, 2022-05-23:
I finally got my SwitchBot cracked open. (I found it wasn't glued shut, and you can jam/hammer a screwdriver to force it open. I've soldered onto the tiny little pads, and esptool can talk to everything! Confirmed on mine also, 4MB flash chip. Interesting, because Tasmota was saying 2MB detected. (It must be the partition confusing it.) I'm going to try and see if I can get a 4MB Tasmota manually flashed, and see if I can get it to accept an OTA after that.
[image]
[image]

Cossid, 2022-05-23:
> I finally got my SwitchBot cracked open. (I found it wasn't glued shut, and you can jam/hammer a screwdriver to force it open. I've soldered onto the tiny little pads, and esptool can talk to [...]

Yep, you will be able to manually flash just fine, and OTA will work fine when you're on the 4M version.  Just a reminder you will need the esp32c3.factory.bin (https://github.com/arendst/Tasmota/releases/download/v11.1.0/tasmota32c3.factory.bin) for manual flashing.  The issue is this tool was using a specific 2M version, and that version does not easily allow further OTA because of space considerations, and that version didn't have the new real flash size check and partition expansion capabilities.  I have a PR in to make this tool correctly use the 4M version moving forward.  I have already fully tested everything.  The only challenge left is to find a way to get the devices that were flashed with the 2M version to update OTA without having to open the devices, otherwise they are permanently stuck on the 2M 11.1.0.1 build.

Yes, I also opened mine with sprudgers and leveraged it open with no damage, but they must be glued back shut because the ultrasonic plastic weld is broken.

kendallgoto, 2022-05-23:
Resolved in #8 & release v1.1.

I'm moving discussion for "re-flashing" the 2MB switches to #9

closed 2022-05-23 by kendallgoto

Jason2866, 2022-05-23:
> Yep, you will be able to manually flash just fine, and OTA will work fine when you're on the 4M version. Just a reminder you will need the esp32c3.factory.bin [...]

I think the 2MB flashed devices are stuck on this version, since writing a valid partition scheme is not possible with the firmware from March.

Cossid, 2022-05-23:
> I think the 2MB flashed devices are stuck on this version, since writing a valid partition scheme is not possible with the firmware from March.

The person I was replying to is doing USB serial flashing to 4M, which will work fine, and once he is on that, OTA will work fine, that is what I was responding to.