kendallgoto/switchbota #15: Switchbot calibration values for Tasmota?

linagee, 2022-05-23:
Not sure if this seems out of scope for this repo, but it's weird that my Switchbot reads 172Vac when I know it's 120Vac coming out of the wall. Not sure if a calibration problem, or are we using the correct power measurement chip? If calibration, can it be saved in the configuration template string?

triggerx, 2022-05-23:
Use the following instructions to calibrate it... always a required first step with Tasmota.  https://tasmota.github.io/docs/Power-Monitoring-Calibration/

kendallgoto, 2022-05-23:
yes - unfortunately, the power chips all require a quick calibration. They're all factory-calibrated but in flashing tasmota, you lose the factory calibration. One plug's calibration does not translate exactly to another plug.

This would be written on the templates page for the switchbot plug, but it's pending an [issue](https://github.com/blakadder/templates/issues/1394) I opened previously

closed 2022-05-23 by kendallgoto

linagee, 2022-05-23:
In case values are similar (or just, better than defaults) for anyone else's SwitchBot W1901400, I used a Brymen BM869s multimeter, P4400 Kill-A-Watt, and a 450W space heater as a resistive load. (Which gave a power factor of 0.97, and there are some electronics in the heater.)

To use my values, open your Tasmota Console and enter:
```
PowerCal 8114
VoltageCal 1384
CurrentCal 3170
```
(Power/Current were from the Kill-A-Watt, and Voltage was from the Brymen multimeter, which was about two volts higher. It seems these values don't get stored in the template string.)