kendallgoto/switchbota #9: Flashing "full" Tasmota from 2MB build

kendallgoto, 2022-05-23:
Release v1.0 of this project flashed plugs onto the ESP32-C3 2MB build of Tasmota. While this version supports core features, such as wifi connectivity and MQTT, it does not support future OTAs and more niche Tasmota features. However, the Switchbot plugs are fully compatible with the "full" 4MB version of Tasmota. Release v1.1 has made the change to the 4MB Tasmota version, as per #7 and #8.

This issue stands as an exploration into the process of upgrading "2MB build" plugs to "4MB builds" by means of out-of-region eeprom writes, by modifying the flash or partition table to re-flash the device manually.

To the best of my knowledge: the existing partition table cannot be overwritten due to the CONFIG_SPI_FLASH_DANGEROUS_WRITE build flag on Tasmota, preventing live modification of the partition table. This flag may be bypassed on the IDF with direct access to the ``esp_rom_`` library, though I am not personally familiar with calling ESP-IDF functions via the Arduino environment, or in general writing Arduino-based ESP images.

Cossid, 2022-05-23:
Precursor info:

Structure as it exists:
1792KB APP0 partition with 1408KB used space, 384KB free space
192KB SPIFFS partition (I'm still not sure of it's purpose/necessity yet) after the APP0 partition

The installed Tasmota build does not support gz for OTA bins, only uncompressed bins.

Additional areas to consider:
- there is a 192KB 'SPIFFS' partition following APP0 on the 2M tasmota partition structure.  I am unsure if it is necessary/required, and might be able to be freed for expanding APP0
- A replacement app.bin in 2M format (Tasmota checks that any uploaded bin matches the current flash size apparently) that is:
    - 384KB or smaller with no adjustments to Tasmota.
    - 512KB or smaller with SPIFFS reduced to 64k (likely the smallest shrink size to to alignment) and APP0 expanded with the difference.
    - 576KB or smaller with the SPIFFS partition removed and APP0 expanded with its full capacity.

Said 2M bin would need 1 partitions, likely equally split at 952KB, and would not allow flashing to full Tasmota, but would probably allow either:
- Reflashing with the original app.bin with correct partition layout
- Flashing Tasmota 4M safeboot and OTAing up to full release

With compile/linking changes, I have managed to get app.bin down to 512KB (but it is completely untested in terms of functionality).  I'm not sure if <384KB is feasible,

triggerx, 2022-07-13:
Hi there any luck on this?  Took the version 1.0 on my first plug, love it so much I bought 3 more and flashed 1.0.  Perhaps I may be fine on that version of Tasmota forever on those 4 plugs, but would like to upgrade them if ever possible.  At $5.25 a pop on Prime Day, I loaded up on more!  Thanks for your work on this project... hands down the best Tasmota-able plug on the market today!

kendallgoto, 2022-07-13:
@triggerx I haven't had much time to really dig into this problem - hopefully soon though. I too have 50+ plugs that I've programmed with the minimal tasmota image that I'd love to migrate. In the meantime, however, those plugs are working great and I can't imagine what feature changes I'd need from future Tasmota builds. My biggest priority was getting future plugs running the full Tasmota version, which luckily we were able to accomplish a few months ago.

Obviously, there's some value in upgrading existing plugs due to security, etc., changes as well. I'm fairly confident that there is a route to upgrade the minimal images, but it requires more work on the Tasmota-side that I personally haven't been able to contribute. I'll share updates when I have more time to do research, but as of now this is an open issue for anyone with the skills to tackle it.