kendallgoto/switchbota #17: Bricked first two of set of four

reedstrm, 2022-06-10:
Whelp, grabbed a set of these, and followed along - was able to do the DNS hijack via my router (openWRT) to point it to my linux laptop, where I ran the firmware server. Had the most trouble getting the app to find the plugs - it'd go into "scanning bluetooth" mode, and not show me the manual button when it didn't find anything, so had to be fast! Paired a couple of my set of four. Both firmware V1.3. I've done some bluetooth hackery before, so used linux bluetooth commands (hcitool and gatttool, specifically) - to write the first value, triggered first download, second write triggered the second download. Then ... no bueno. Switch reset, dropped off the network, dropped bluetooth - bricked. Hashes of firmware files used:

2d62194a6b11827de41c9d06b3da0d3cbfdbc824b254eb79a8238f557e21a2f8  app.bin
84fe8ee7960fa0e29aad5895c16c5fa1bfd94b819d4ca0fe8701b74593b90a86  payload.bin

I did notice that the DNS record for the switch's IP changed names, not sure exactly when, from something like "sbotmini-86"  to "espressif". I'm going to crack these open to get to the serial pins to reflash, but before I do, is there anything you'd like me to try to help diagnose what may have happened? I'll dump whatever is in the flash before overwriting it, at minimum. I've got two more chances. :)

(PS, I've found wireshark can capture bluetooth via an android phone's adb HCI snoop interface, so I'm going to attempt to capture the initial bluetooth/wifi setup, to hopefully be able to bypass using switchbot's app at all)

kendallgoto, 2022-06-10:
Sorry to hear your devices didn't smoothly flash! Obviously, that's not the desired behavior I'd want for anyone.

A quick clarification: the devices will not rejoin your network or broadcast on bluetooth after flashing. They create a new Tasmota network that you must join to configure the device. Have you checked for this network? This is the default Tasmota behavior but it's unclear from your message if you looked, or were just expecting the plugs to reconnect to the previous network after flashing. This behavior might not be intuitive if you aren't familiar with Tasmota & may need to be better documented on our end. 

Were your plugs flashed with V1.3 from the factory? Or did you update them in the app prior to flashing? I haven't personally ran v1.3 myself, so I'm not sure of the differences / if they may have caused an issue.
 
The dns name being espressif would likely just be the device name that it identifies with in the intermediate SwitchbOTA app - nothing special there I don't think. 

With regard to bluetooth sniffing: I did a good amount of sniffing already and documented some here https://github.com/kendallgoto/switchbota/issues/3#issuecomment-1121864522 - although, mostly to trigger the updates. I still use the app to configure the device's network connectivity, etc. It would be great to get that provisioning process documented more if possible!

reedstrm, 2022-06-10:
Yup, purely a case of me staying up too late and not thinking all the way o the end - Thanks! For my penance, I'll try hard on the document/reverse engineer the initial provisioning.

closed 2022-06-10 by kendallgoto