kendallgoto/switchbota #21: Bricked

euphguy, 2022-07-17:
Well I think I bricked one of my plugs. I set up the server and the dns redirect. I sent the Bluetooth code. It dropped off the network and never returned. 

Now the plug seems dead. No led. No response from the button or click from the relay. 

I may just return it unless you have any suggestions.

kendallgoto, 2022-07-17:
did you look for a new wifi network after plugging it into the wall? if you're not familiar with tasmota: after the OTA process, the plugs don't rejoin your network automatically. They create a new network that you must connect to and repair it.

euphguy, 2022-07-17:
I did, I do not see any new network. Tasmota or otherwise.

kendallgoto, 2022-07-17:
That's disappointing to hear - sorry about that! Did you have other plugs that worked out?

The unfortunate part of OTAs like this is that if anything goes wrong, they're not recoverable & it's almost impossible to diagnose what went wrong.

Do you happen to have the logs from the Node server still? They should show a few requests from the device in the history.

euphguy, 2022-07-17:
This was my first attempt. The plugs were a Prime Day impulse buy and an easy return. 

Thanks for the quick response!

triggerx, 2022-07-18:
Keep trying euphguy.... I could have sworn I bricked one of mine, but I kept trying and it ended up working out perfectly.  Twice I had to use the bluetooth code, and one of those times I had to try several times, but like kendallgoto said, eventually I saw the Tasmota wifi AP magically come up.  Try again!

closed 2022-07-18 by kendallgoto