kendallgoto/switchbota #22: Middleware firmware requirement

Cossid, 2022-07-19:
I was wondering if you could provide some insight on what the reasoning is for requiring an app.bin middleware firmware?  I'm assuming it might have just been the first thing you tried, or you preferred a different partition layout or you were worried about insufficient partition space for using a .factory. bin?

In testing, it appears it can be skipped by just sending a regular non-factory tasmota bin file (which fits within the app partitions) on the first request instead of the app.bin middleware.  I have tried it on a couple devices, and so far, everything seems fine.  Am I missing something?

kendallgoto, 2022-07-19:
my recollection is, unless something has changed on the Tasmota side, that Tasmota utilizes the modified ESP32 Arduino-based bootloader + app images which do not natively work with the standard Espressif ESP-IDF images (in the platformio config for Tasmota, that's defined as `arduino` instead of `espidf`).

When I first tested SwitchbOTA, the on-board bootloader on the plugs certainly wasn't able to run Tasmota app images for this reason, which is why I reflash the entire "factory" bin, overwriting the bootloader, partition table, and app images with SwitchbOTA. Certainly, I'd expect the partition table at the very least to differ between Tasmota and Switchbot's native firmware, causing issues.

I'd have to revisit it on my end, but I know I wasn't able to reproduce what you are saying back in April when I first got my Switchbot plugs. There's also some further discussion [here](https://www.esp32.com/viewtopic.php?t=22639) related to the difference between the Arduino and ESP-IDF bootloader. It's also the reason why the Sonoff OTA program, [Espressif2Arduino](https://github.com/khcnz/Espressif2Arduino) exists, which is similar to SwitchbOTA, just built for Sonoff devices. If the middle step firmware isn't required, certainly I'd be happy to provide SwitchbOTA as solely a DNS proxy to deliver the Tasmota payload, but that hasn't been my experience.

Cossid, 2022-07-19:
As I mentioned previously, I have lost my original OEM plug firmware due to an accidental overwrite, so my initial testing was the against the firmware I re-created using instructions you gave me (if you wouldn't mind, please upload an oem plug bin dump so I can better test).  However, I just tried it on a SwitchBot Color Bulb with OEM firmware, and it seemed to work fine, though I'm still exploring any potential issues.

Here was my process for a new bulb:

- Modify server to download https://github.com/arendst/Tasmota/releases/download/v11.1.0/tasmota32c3.bin and instead of serving app.bin, I served that file instead
- Request a firmware update through the app
- Tasmota 11.1.0 is flashed to the bulb and loads in AP mode.
- Connect to AP, enter wifi credentials and it reboots
- OTA to 12.0.2 via interface, and it works without issue (though web server takes a while to load, this is a general ESP32-C3 issue).

This does leave the stock partition layout, which is slightly different than Tasmotas, but it appears to work so far.  Tasmota did change partition layout options in v12, so I'm looking to see what it takes to switch between them, but being as I started with v11, it doesn't appear related to any of the partition changes in v12.

Cossid, 2022-07-19:
First issue (as expected) is there is no partition for FileSystem, so Berry .tapps cannot be used (which is the current method for changing partition layouts in Tasmota v12, slight irony!)

kendallgoto, 2022-07-19:
Thank you for this information. That's very interesting - I would not have expected that to work out of the box + it didn't seem to work in my original testing, though perhaps another issue was confounding the problem.

Regarding the partition table, this is the current one being used on the devices
Plug Mini (original + HomeKit versions):
```
phy_init,data,phy,0x9000,4K,
otadata,data,ota,0xa000,8K,
keydata,data,254,0xc000,8K,
nvs,data,nvs,0xe000,24K,
region,data,253,0x14000,4K,
factory_nvs,data,nvs,0x15000,16K,
energydata,data,fat,0x19000,24K,
ota_0,app,ota_0,0x20000,1904K,
ota_1,app,ota_1,0x220000,1904K,
```

Compared to Tasmota's factory partition table:
```
nvs,data,nvs,0x9000,20K,
otadata,data,ota,0xe000,8K,
app0,app,ota_0,0x10000,1856K,
app1,app,ota_1,0x1e0000,1856K,
spiffs,data,spiffs,0x3b0000,320K,
```

I would think that these would differ too much, preventing the Tasmota binaries from appropriately being able to address the NVS partition to save settings, etc. without the forceful bootloader+partition table changes that I make. Evidently, your findings differ. I'll have to dig out a new plug + make a dump of its stock firmware - let me look further into this.

Cossid, 2022-07-19:
Yeah, I'm not pushing to do this one way or the other, just looking for insight/discussion.

There are pros and cons of doing it each way.

Pros for going straight to Tasmota without intermediate:

- Possibly easier/less error prone (one fewer blind flashes)
- Would work 'well enough' for most users
- Potential to read stock nvs data like calibration info to be saved/re-used instead of manual calibration later

Cons for going straight to Tasmota without intermediate:

- Non-standard partition layout
- No FileSystem partition

Pros for using intermediate firmware:

- Standard partition layout, including a FileSystem partition
- Possibly more resilient against failure with a fall-back mode

Cons for using intermediate firmware:

- Extra steps/flashing is more prone to misunderstandings of how the device should be responding and certain stages
- Loss of all factory data at the very first stage (relatively minor)
- Currently ties to specific md5 hashed versions of Tasmota (obviously you can upgrade after any version of tasmota is loaded, just a potential maintenance con [I bring this up because Switchbot Bulbs require a newer version which doesn't have an official release yet.  I will probably create a PR when there is a stable release bin for it.])

In fact, in going straight to Tasmota and discovering the partition layout issue of lacking FileSystem, I just simply flashed app.bin and let it go back through the normal procedure to re-init the partitioning (unfortunately it's not that simple on the 2M issue due to space constraints).

I honestly think keeping the middleware is probably the best choice, but it will potentially be more complex in expanding this beyond ESP32-C3 (for example, Switchbot makes a humidifier which is full ESP32-WROOM-32D (not C3) that appears to use the same process which would require a separate app.bin)

kendallgoto, 2022-07-19:
As you described, I performed the OTA process. Interestingly, the device did come up, however the UART logs off the device seem to indicate that Tasmota is not behaving correctly. After powering on, the device threw a core panic several times but eventually came up on its own (I'm not very familiar with Tasmota in this way - maybe it has some method of "safe booting" after failed attempts).
For completeness: https://pastebin.com/ZyqEMEWY
Note that some of the Python errors included here is probably because I'm monitoring with `idf.py` and don't have the debug symbols for Tasmota loaded - nevertheless, the device is throwing core panics on boot. It seems to do this every boot.

I connected it to my network and it joined successfully. I loaded my template from the Tasmota Template DB + that seemed to come up just fine as well. It did seem like it persisted data throughout reboots, etc., as well, although still panicking every boot.
```json
{"Info3":{"RestartReason":{"Exception":5,"Reason":"Load access fault","EPC":"4205a624","EXCVADDR":"00000000","CallChain":["4207985e","4207980e","4207985e","4206b058","4206ee4a","4207980e","420712e0","42069e84","4207980e","42069fde","4206ee4a","4206f2ee","42065832","4206af7c","4207106c","42067a20","42068f54","4206ee4a","4206f2ee","42067bda","42067c40","42067bca","42067c36","4038a9de","4038aa1e","4038a9de","4038a9de","4038aa1e","4206ebb8","4206ec28","40382a0e","4206ec18","4206a048","4206a17c","42067dec","42067a20","42065884","42020c04","42043b7c","42088000","4208c3ce","403890f4"]},"BootCount":6}}
```
Every time it comes up, it notes `Some settings have been reset (4)`

In Tasmota, I ran the OTA process to v12.0.2 and that worked OK, but still it was panicking 6+ times every reboot. At some point, it actually started to panic so many times on bootup that it would wipe my Tasmota settings every time, presumably due to the plug-unplug-repeat recovery loop built into Tasmota. 

Are you seeing the same kernel panics? They should show up in the Tasmota logs right after boot, even if you don't have UART access:
```
00:00:00.000 HDW: ESP32-C3 
00:00:00.005 CFG: Loaded from NVS, Count 51
00:00:00.007 FRC: Some settings have been reset (4)
06:02:35.596 RSL: INFO3 = {"Info3":{"RestartReason":{"Exception":5,"Reason":"Load access fault","EPC":"4205f30a","EXCVADDR":"00000000","CallChain":["4207b898","4207b848","4207b898","4206cce8","42070b90","4207b848","4207303a","4206bae6","4207b848","4206bc40","42070b90","42071048","420672d0","4206cbde","42072dc6","42069676","42070fb4","42069830","4206abb6","42070b90","42071048","42069830","42069896","42069820","4206988c","4038be22","4038be62","4038be22","4038be22","4038be62","40383448","4038345c","4206c644","4206a25e","42070efc","4005890e","4038a660","42069a42","42069676","42067322","42021c7e","42044d16","42089000","4208cbe4","40389d96"]},"BootCount":15}}
```

Personally, I use Tasmota to serve power logs to my own MQTT server. I was able to get that still working, so that's OK at least. As you noted, it didn't seem like Berry worked, although I'm not very familiar with it.

Overall: my impression is that this does indeed work - much to my surprise. But, as you've noted, I think there's still value + long term stability added by the, albeit risky, one time OTA process.

kendallgoto, 2022-07-19:
> Pros for going straight to Tasmota without intermediate:
> * Possibly easier/less error prone (one fewer blind flashes)
> * Would work 'well enough' for most users
> * Potential to read stock nvs data like calibration info to be saved/re-used instead of manual calibration later

Agree here. Although, we could read stock NVS calibration from the base device within SwitchbOTA and write it into NVS after we flash. I didn't really explore this because I didn't know how the original calibration data was stored + how it would get stored in Tasmota. Without much experience on either code bases, I assumed it wasn't just a set of numbers I could easily carry over in NVS.

> Cons for using intermediate firmware:
> * Extra steps/flashing is more prone to misunderstandings of how the device should be responding and certain stages
> * Loss of all factory data at the very first stage (relatively minor)
> * Currently ties to specific md5 hashed versions of Tasmota
I agree with these concerns as well. Blindly flashing the device _is_ inherently risky and it's a bit of a "black box" while the process is going on. I don't see the MD5 pinning to be a specific concern - there's a few ways we could automate getting around that without too much issue. For the moment, it's hardcoded because we can, generally, always OTA to a given Tasmota version and then just upgrade regularly from there, even if our base tasmota version is years old. Obviously if this project was adapted to other Switchbot / ESP-based devices not built on the C3, there would have to be some changes. 

Moving forward: I think we can perhaps document both paths and add a switch in the node server to bypass the middleware firmware. Though, I'd personally still push for the "proper" bootloader/partition table/etc with SwitchbOTA for better confidence in long-term stability.

Cossid, 2022-07-19:
Honestly, I wasn't paying much attention to the logs or looking for panics.  WebServer always seems to have general issues with WebServer on ESP32-C3 when booting up, so I attributed any wonkyness to that without it registering in my mind that it may have been worse than normal.  Tasmota does have a recovery mode, somewhere between 6 and 8 reboots I believe, but I'm also not a Tasmota guru.  I don't currently have a device in the 'straight-to-tasmota' state to re-check, though I may do so later, I have several bulbs that are still stock that I am keeping that way to potentially use to improve this process where it can be, so once I have a clearer plan, I may start doing that (but having a stock plug firmware would help the most still I think because that is the easiest for me to serial flash still).

Yes, my preference is still to keep the middleware as well, it provides the most flexibility, and the cons are not that significant.

As part of removing some of the 'blindness' I'm going to add a simple PR that will add just a touch more logging for the server that will notify when a download has completed, which will hopefully help prevent people from impatiently rebooting too early.

I think this discussion has gone far enough, and it will be here for anybody else curious to revisit.  I will close this issue out unless someone deems more discussion is necessary.

closed 2022-07-19 by Cossid