kendallgoto/switchbota #23: Unreleased SwitchBot Firmware Documentation

kendallgoto, 2022-07-19:
I haven't studied any of these binaries, so take care:

Debug v1.0 http://www.wohand.com/version/wocaodebug/firmware/WoPlugUS/WoPlugUS_V10.bin (unreleased)
Debug v1.1 http://www.wohand.com/version/wocaodebug/firmware/WoPlugUS/WoPlugUS_V11.bin
Debug v1.2 http://www.wohand.com/version/wocaodebug/firmware/WoPlugUS/WoPlugUS_V12.bin
Debug v1.3 http://www.wohand.com/version/wocaodebug/firmware/WoPlugUS/WoPlugUS_V13.bin
Debug v1.4 http://www.wohand.com/version/wocaodebug/firmware/WoPlugUS/WoPlugUS_V14.bin
Debug v1.5 http://www.wohand.com/version/wocaodebug/firmware/WoPlugUS/WoPlugUS_V15.bin (unreleased 2022-05-06)

Production v1.1 http://www.wohand.com/version/wocaotech/firmware/WoPlugUS/WoPlugUS_V11.bin
Production v1.2 http://www.wohand.com/version/wocaotech/firmware/WoPlugUS/WoPlugUS_V12.bin (latest for regular plugs as of 2022-07-19)
Production v1.3 http://www.wohand.com/version/wocaotech/firmware/WoPlugUS/WoPlugUS_V13.bin 
Production v1.4 http://www.wohand.com/version/wocaotech/firmware/WoPlugUS/WoPlugUS_V14.bin (latest for HomeKit plugs as of 2022-07-19)

Just wanted to document the links I found - I'm not sure what the practical difference is between the debug and production firmware versions in each case.

Cossid, 2022-07-19:
Curious, are there any hardware differences for the HomeKit plugs, or is it just software?

Additional hardware:

Bulbs:
Debug v1.0 http://www.wohand.com/version/wocaodebug/firmware/WoBulb/WoBulb_V10.bin
Debug v1.1 http://www.wohand.com/version/wocaodebug/firmware/WoBulb/WoBulb_V11.bin
Debug v1.2 http://www.wohand.com/version/wocaodebug/firmware/WoBulb/WoBulb_V12.bin
Debug v1.3 http://www.wohand.com/version/wocaodebug/firmware/WoBulb/WoBulb_V13.bin
Debug v1.4 http://www.wohand.com/version/wocaodebug/firmware/WoBulb/WoBulb_V14.bin
Debug v1.5 http://www.wohand.com/version/wocaodebug/firmware/WoBulb/WoBulb_V15.bin
Debug v1.6 http://www.wohand.com/version/wocaodebug/firmware/WoBulb/WoBulb_V16.bin
Debug v1.7 http://www.wohand.com/version/wocaodebug/firmware/WoBulb/WoBulb_V17.bin

Production v1.0 http://www.wohand.com/version/wocaotech/firmware/WoBulb/WoBulb_V10.bin
Production v1.0 http://www.wohand.com/version/wocaotech/firmware/WoBulb/WoBulb_V11.bin
Production v1.0 http://www.wohand.com/version/wocaotech/firmware/WoBulb/WoBulb_V12.bin
Production v1.0 http://www.wohand.com/version/wocaotech/firmware/WoBulb/WoBulb_V13.bin
Production v1.0 http://www.wohand.com/version/wocaotech/firmware/WoBulb/WoBulb_V14.bin
Production v1.0 http://www.wohand.com/version/wocaotech/firmware/WoBulb/WoBulb_V15.bin
Production v1.0 http://www.wohand.com/version/wocaotech/firmware/WoBulb/WoBulb_V16.bin
Production v1.0 http://www.wohand.com/version/wocaotech/firmware/WoBulb/WoBulb_V17.bin

Humidifier:
Debug v1.0 http://www.wohand.com/version/wocaodebug/firmware/WoHumi/WoHumi_V10.bin
Debug v1.1 http://www.wohand.com/version/wocaodebug/firmware/WoHumi/WoHumi_V11.bin
Debug v1.2 http://www.wohand.com/version/wocaodebug/firmware/WoHumi/WoHumi_V12.bin
Debug v1.3 http://www.wohand.com/version/wocaodebug/firmware/WoHumi/WoHumi_V13.bin
Debug v1.4 http://www.wohand.com/version/wocaodebug/firmware/WoHumi/WoHumi_V14.bin
Debug v1.5 http://www.wohand.com/version/wocaodebug/firmware/WoHumi/WoHumi_V15.bin
Debug v1.6 http://www.wohand.com/version/wocaodebug/firmware/WoHumi/WoHumi_V16.bin
Debug v1.7 http://www.wohand.com/version/wocaodebug/firmware/WoHumi/WoHumi_V17.bin
Debug v1.8 http://www.wohand.com/version/wocaodebug/firmware/WoHumi/WoHumi_V18.bin
Debug v1.9 http://www.wohand.com/version/wocaodebug/firmware/WoHumi/WoHumi_V19.bin
Debug v2.0 http://www.wohand.com/version/wocaodebug/firmware/WoHumi/WoHumi_V20.bin

Production v1.4 http://www.wohand.com/version/wocaotech/firmware/WoHumi/WoHumi_V14.bin
Production v1.5 http://www.wohand.com/version/wocaotech/firmware/WoHumi/WoHumi_V15.bin
Production v1.6 http://www.wohand.com/version/wocaotech/firmware/WoHumi/WoHumi_V16.bin
Production v1.7 http://www.wohand.com/version/wocaotech/firmware/WoHumi/WoHumi_V17.bin
Production v1.8 http://www.wohand.com/version/wocaotech/firmware/WoHumi/WoHumi_V18.bin
Production v1.9 http://www.wohand.com/version/wocaotech/firmware/WoHumi/WoHumi_V19.bin

evlo, 2022-09-02:
So with this would it be possible to just upload it to the device through switchbota and revert to factory?
I'm asking as ie. sonoff has different firmware for each unit.

kendallgoto, 2022-09-02:
no - these are OTA images. They only contain the app-code portion of the app. Switchbota writes the entire internal flash of the device, requiring the bootloader, partition table, nvs values, etc.

You can, however, trigger one of these to install using the original firmware and a custom crafted BLE message. For the Switchbot Plugs, I wrote the BLE info here:
https://github.com/kendallgoto/switchbot-plug-influxdb/blob/master/notes.txt#L47

To install a debug build on a production plug, you'd need to likely spoof the web server using something like Switchbota's node webserver (with modifications) to serve the Debug build for a Production request.

kendallgoto, 2023-03-30:
Updated OP w/ new releases I saw on 3/29/2023, including:
1.5 has been released to production from debug
1.6 has appeared in debug

AJolly, 2024-05-24:
I can confirm this worked successfully with 1.5, including letting me flash to esphome.

smmille1, 2025-10-21:
I recently got a SwitchBot Relay Switch 1 (twin pack, actually). It's interestingly using slightly different paths.
Current firmware link: http://www.wohand.com/version/wocaotech/firmwareV2/W1083002_wifi/WoSwitch_V16.bin

I pulled this out of the binary. Nulls have been replaced with line breaks. It references several models:
```
W1083000

WoSwitch1PM
http://s3.ap-northeast-2.amazonaws.com/www.wohand.com-test/version/wocaotech/firmwareV2/W1083000_wifi/WoSwitch_V

http://www.wohand.com/version/wocaotech/firmwareV2/W1083000_wifi/WoSwitch_V
W1083001

WoSwitch2PM
http://s3.ap-northeast-2.amazonaws.com/www.wohand.com-test/version/wocaotech/firmwareV2/W1083001_wifi/WoSwitch_V

http://www.wohand.com/version/wocaotech/firmwareV2/W1083001_wifi/WoSwitch_V
W1083002

WoSwitch1Plus

http://s3.ap-northeast-2.amazonaws.com/www.wohand.com-test/version/wocaotech/firmwareV2/W1083002_wifi/WoSwitch_V

http://www.wohand.com/version/wocaotech/firmwareV2/W1083002_wifi/WoSwitch_V
WoSwitchGDO
http://s3.ap-northeast-2.amazonaws.com/www.wohand.com-test/version/wocaotech/firmwareV2/W1083003_wifi/WoSwitch_V
```

The garage door opener (presumably WoSwitchGDO) looks to be the same hardware as the Relay Switch 1, but it must use slightly different firmware as I can only register the one I have as a Relay Switch even if I wire it per the garage door opener instructions. I ultimately want to use one as a garage door opener so I'm tinkering to see if I want to try to find their GDO firmware or just try and switch to Tasmota. I'm guessing all the other version URLs were identified by trial and error (i.e. there isn't a way to ask for a list)? I have yet to start trying other URLs or crack it open to look at the board.