kendallgoto/switchbota #29: Firmware 1.3

BuxtonCalvin, 2022-08-08:
I couldn't get the tasmota update to work with the latest switchbota firmware (1.3).  The JS server works fine, but seems to hang when I open the switchbota android app:

[image]

And I can't pair the plug with native Bluetooth, so I can't use software to send Bluetooth commands to start the upgrade process.  The switchbota android app connects both with wifi and Bluetooth.

kendallgoto, 2022-08-08:
if you can't connect to the plug at all over bluetooth, you'll need to work that out between your phone / the plug / switchbot. if you're on the latest firmware, you'll need to pair the plug (i.e. connect it to your wifi network) and then send the Bluetooth signal via nRF Connect. There's no other way to install the firmware from switchbota.

BuxtonCalvin, 2022-08-08:
Ok thanks for that.  Which app within the "nRF Connect" suite is the one I should use to send the Bluetooth signal.  The apps I tried so far within the suite, all request a development server of some sort, to initialize.  I downloaded/using the windows version of the software.

kendallgoto, 2022-08-08:
sorry, i was referring to the nRF connect mobile application. their desktop app is pretty different. If you can't get it working on your phone, you can try the lightblue app on windows, but I haven't personally tried it so I can't offer much guidance. Presumably, within Lightblue, you can connect to the BLE device and send the signal as described in #3

BuxtonCalvin, 2022-08-08:
Ok, with your help, I got it to work with the android version of the nRF connect software.   One thing I would add to #3 topic is that once you see the "/payload.bin" on the server console, go into your phone and look for the tasmota SSID.  Once you connect to that SSID, you can change the WiFi to your local network.  I had forgotten about that step from other Tasmota installs, and was freaking out that I had bricked the device as it was completely unresponsive.

Also, the direction you gave to "reset" the device is a little vague.  I sent the the second hex command about a minute after the first without doing anything else to the device, as I wasn't sure what you meant.....

BuxtonCalvin, 2022-08-08:
Is it possible to swap out the payload.bin with another tasmota firmware?  I wanted to see if I could flash the bluetooth version of the esp32 firmware.

Thanks for all your help with this.

kendallgoto, 2022-08-08:
if you're already on tasmota, just perform an OTA within Tasmota to select the desired build - switchbota doesn't really matter at that point.

With that said, you are limited to the ESP32-C3 builds that are releases unless you build on from source. Those are here: https://github.com/arendst/Tasmota-firmware/tree/main/release-firmware/tasmota32. The "ESP32" builds won't work, since it's a different architecture. Only the ones labeled for the "ESP32 C3".

BuxtonCalvin, 2022-08-08:
Yeah, I tried the ESP32 Bluetooth and the update failed.  What's odd is that if you set ESP32 C3 as the filter in the Tasmota web serial flasher, you get a list of supposedly compatible firmwares.  The Bluetooth compile is listed as compatible in the drop down menu.  I could compile an ESP32 C3 version with Bluetooth in VSC as there's only one switch to set in the compile instructions, but I fear I'd be getting too far into the weeds, as if anything failed, I'd be lost.  Hopefully, the Tasmota project managers will see a similar compile as deserving its own binary as I'm sure many will want the BLE functionality.  In my case, that's why I bought these particular switches as I need to read BLE sensor data.
Thanks again for your help and the great work on this type of switch.  I'll close the PR out as the work-around solved the install problem.

closed 2022-08-08 by BuxtonCalvin

kendallgoto, 2022-08-08:
this binary will probably do the trick:
https://github.com/tasmota/install/blob/main/firmware/unofficial/tasmota32c3-bluetooth.bin
(for OTA from Tasmota)

BuxtonCalvin, 2022-08-09:
OMG dude! I spent a couple hours searching for something similar with no joy. Including searches on the site!  The binary works!   Thank you so much.