kendallgoto/switchbota #31: Update Tasmota FW to latest version

kendallgoto, 2022-08-21:
Now that Tasmota officially supports the ESP32-C3 (#30), the currently installed Tasmota version `https://github.com/arendst/Tasmota/releases/download/v11.1.0/tasmota32c3.factory.bin` should get bumped to be the stable, official build `http://ota.tasmota.com/tasmota32/release-12.1.0/tasmota32c3.factory.bin` and tested.

We could also pull from `tasmota32/latest/` directly, but this would require redesigning our MD5 verification system, as this requires knowing the FW MD5 dynamically. However, in the short term, it's likely safer to just default to 12.1.0 and allow for users to OTA upwards from there.

jamorin, 2022-08-30:
thanks for the awesome project. I've noticed a network reconnection issue with my plugs using the latest ota firmware, 12.1.1 at the time of writing. Actually, 12.1.0 and 12.0.2.4 have issue as well.

For example. If I reboot my home router, afterwards the plug will not reconnect to network indefinitely. I have to hard power cycle the plug to reconnect. 

I have downgraded to version 12.0.1.2 which so far is working fine https://github.com/tasmota/install/blob/801bde23cc440594292ddb9da207388982cc6d36/firmware/unofficial/tasmota32c3-bluetooth.bin

kendallgoto, 2022-08-30:
thanks for noting this & testing it out. i'm not personally in a big rush to push us to the "official" tasmota release since users can freely OTA, but long-term hopefully the esp32-c3 related quirks on the official release get worked out and we can move over.

Cossid, 2022-12-04:
Word of caution against this, Tasmota switched to a partition layout for factory images starting with 12.0 that is incompatible with ESPHome.  See https://tasmota.github.io/docs/Safeboot/ for more information.

I would strongly recommend remaining at 11.x for the first flash.  You can always upgrade to 12 afterwards OTA without any issue.  But if you start with 12.x, ESPHome is no longer an option without a serial flash.

kendallgoto, 2022-12-05:
also good to know! thank you for sharing... definitely not an edge use case i'd be aware of personally.