kendallgoto/switchbota #36: Relay turns off when Tasmota reboots

psitem, 2022-11-20:
I just converted a bunch of the mini plugs and noticed that the relay doesn't stay on while Tasmota restarts (warm boot). Having `PowerOnState 3` restores the state but having attached devices power cycle because Tasmota rebooted is undesirable. 

Checking other Tasmota devices I have, my "SWB1" power strips and Sonoff S31s maintain the relay state during a reboot but none of my light switches do.

Is this a hardware limitation? A config option I haven't been able to find? Something else?

I'm using my own Tasmota build to have Bluetooth/BLE/Berry support but I've also tried the official release tasmota32c3.bin and a few others.

<details>
<summary>Status0 details:</summary>

```
{
   "Status":{
      "Module":0,
      "DeviceName":"Tasmota",
      "FriendlyName":[
         "Tasmota"
      ],
      "Topic":"tasmota_ADCA60",
      "ButtonTopic":"0",
      "Power":1,
      "PowerOnState":3,
      "LedState":1,
      "LedMask":"FFFF",
      "SaveData":1,
      "SaveState":1,
      "SwitchTopic":"0",
      "SwitchMode":[
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0
      ],
      "ButtonRetain":0,
      "SwitchRetain":0,
      "SensorRetain":0,
      "PowerRetain":0,
      "InfoRetain":0,
      "StateRetain":0,
      "StatusRetain":0
   },
   "StatusPRM":{
      "Baudrate":115200,
      "SerialConfig":"8N1",
      "GroupTopic":"tasmotas",
      "OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32c3.bin",
      "RestartReason":"Software reset digital core",
      "Uptime":"0T00:01:24",
      "StartupUTC":"2022-11-20T16:25:56",
      "Sleep":50,
      "CfgHolder":4617,
      "BootCount":7,
      "BCResetTime":"2022-11-20T17:16:02",
      "SaveCount":21
   },
   "StatusFWR":{
      "Version":"12.2.0(ble-berry)",
      "BuildDateTime":"2022-10-27T04:19:26",
      "Core":"2_0_5",
      "SDK":"v4.4.3.4",
      "CpuFrequency":160,
      "Hardware":"ESP32-C3 rev.3",
      "CR":"398/699"
   },
   "StatusLOG":{
      "SerialLog":2,
      "WebLog":2,
      "MqttLog":0,
      "SysLog":0,
      "LogHost":"",
      "LogPort":514,
      "SSId":[
         <removed>
      ],
      "TelePeriod":300,
      "Resolution":"558180C0",
      "SetOption":[
         "00008009",
         "2805C80001000600003C5A0A192800000000",
         "00000248",
         "00006000",
         "00004000",
         "00000000"
      ]
   },
   "StatusMEM":{
      "ProgramSize":1688,
      "Free":1856,
      "Heap":142,
      "StackLowMark":21,
      "PsrMax":0,
      "PsrFree":0,
      "ProgramFlashSize":4096,
      "FlashSize":4096,
      "FlashChipId":"164020",
      "FlashFrequency":80,
      "FlashMode":"DOUT",
      "Features":[
         "00000809",
         "9F9AD7CF",
         "0015A001",
         "B7F7BFCF",
         "05DA9BC0",
         "E0360DC7",
         "000840D2",
         "20200000",
         "D434482D",
         "00000001"
      ],
      "Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,62,63,79,86,87",
      "Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,127"
   },
   "StatusNET":{
      <removed>
   },
   "StatusMQT":{
      "MqttHost":"",
      "MqttPort":1883,
      "MqttClientMask":"DVES_%06X",
      "MqttClient":"DVES_ADCA60",
      "MqttUser":"DVES_USER",
      "MqttCount":0,
      "MAX_PACKET_SIZE":1200,
      "KEEPALIVE":30,
      "SOCKET_TIMEOUT":4
   },
   "StatusTIM":{
      "UTC":"2022-11-20T16:27:20",
      "Local":"2022-11-20T17:27:20",
      "StartDST":"2022-03-27T02:00:00",
      "EndDST":"2022-10-30T03:00:00",
      "Timezone":"+01:00",
      "Sunrise":"08:07",
      "Sunset":"17:04"
   },
   "StatusPTH":{
      "PowerDelta":[
         0,
         0,
         0
      ],
      "PowerLow":0,
      "PowerHigh":0,
      "VoltageLow":0,
      "VoltageHigh":0,
      "CurrentLow":0,
      "CurrentHigh":0
   },
   "StatusSNS":{
      "Time":"2022-11-20T17:27:20",
      "ENERGY":{
         "TotalStartTime":"2022-11-20T17:16:02",
         "Total":0.003,
         "Yesterday":0.000,
         "Today":0.003,
         "Power":1,
         "ApparentPower":2,
         "ReactivePower":2,
         "Factor":0.47,
         "Voltage":168,
         "Current":0.014
      }
   },
   "StatusSTS":{
      "Time":"2022-11-20T17:27:20",
      "Uptime":"0T00:01:24",
      "UptimeSec":84,
      "Heap":141,
      "SleepMode":"Dynamic",
      "Sleep":50,
      "LoadAvg":19,
      "MqttCount":0,
      "Berry":{
         "HeapUsed":4,
         "Objects":49
      },
      "POWER":"ON",
      "Wifi":{
         <removed>
      }
   }
}
```

</details>

kendallgoto, 2022-11-21:
I've always seen this behavior even on official Switchbot FW for these devices. I think it's a hardware limitation. As far as I'm aware, its intentional behavior to ensure the system fails closed if there's a power event / tampering / etc.

I agree that it isn't user-friendly behavior though. I had some plugs resetting in Tasmota on random intervals due to an upstream MQTT issue - appliances throughout my home were randomly coming on and off and I couldn't figure out why.

I'm not sure how Tasmota could bypass this aside from retaining the ESP32-C3 GPIO pin values during the reset, which I'm not entirely sure the esp32s are capable of doing (beyond deep-sleep). The Switchbot plugs don't use a `latching relay` so they need a driving signal or they reset. Maybe the Sonoff plugs use a latching relay or have a capacitor to keep the GPIO signal alive during a reboot, but a quick glance at the Tasmota source code seems to suggest that there's no logic to ensure the GPIOs keep their state during a reboot ... could be wrong about that though.

In any case - this would end up being an upstream issue with Tasmota's codebase.

psitem, 2022-11-21:
Thanks for the response and this project. I think I will raise an issue upstream. I checked a couple more devices and the common factor I'm seeing is the ESP32-C3 — everything I have with a -C3 resets the GPIO state during a reboot while all my ESP8266 devices hold state (at least on the pins configured for relays).

closed 2022-12-14 by kendallgoto