kendallgoto/switchbota #39: Revert to SwitchBot Firmware

donavanbecker, 2023-04-04:
Is there a way to revert back to the factory firmware?

donavanbecker, 2023-04-04:
I went to ESPHome from Tasmota, is there an bin file that I can use to go back to Tasmota?

kendallgoto, 2023-04-04:
i'm not familiar with esphome at all. the `payload.bin` is the esp32c3 factory file from tasmota (https://github.com/arendst/Tasmota/releases/download/v11.1.0/tasmota32c3.factory.bin). presumably you'd not want the factory file, you'd want a single app image, which would be https://github.com/arendst/Tasmota/releases/download/v11.1.0/tasmota32c3.bin. But it would depend on how you went from Tasmota->ESPHome and how you intend to go back ESPHome->Tasmota.

for clarity: a factory image includes the entire system partition table, bootloader, etc and is usually flashed directly to a flash chip. an app image is just a single partition in that partition table that contains the actual app. typical app-level updates use the app image whereas the factory image is (ideally) only flashed once ever.

Switchbota doesn't really cover this topic at all - it's only concerned with moving you from switchbot's factory FW to tasmota. Once you're running Tasmota, you're just like everyone else -- i.e., articles like https://siytek.com/esphome-vs-tasmota/#ESPHome-to-Tasmota would apply.

donavanbecker, 2023-04-04:
> presumably you'd not want the factory file, you'd want a single app image, which would be https://github.com/arendst/Tasmota/releases/download/v11.1.0/tasmota32c3.bin.

This worked!

closed 2023-04-04 by donavanbecker