kendallgoto/switchbota #40: Stream a backup of current firmware over the air before flashing

kendallgoto, 2023-04-04:
It would be a neat feature to stream a bit-for-bit backup of the existing installed FW .bin file and upload it to the switchbota web server before downloading and installing tasmota. This offers the opportunity to restore the original device FW in full, with the device specific calibration, provisioning, etc data.

AJolly, 2025-04-11:
Is it currently possible to restore the original FW? I've had a bunch of these have the relay die on me, and would like to reflash for warantee service

kendallgoto, 2025-04-16:
theoretically, yes. without a backup and without opening it, I don't think there's a documented way at the moment.

AJolly, 2025-04-17:
Is there a documented way to back up the firmware without opening it up? I have a bunch of unflashed switchbots ready to go

kendallgoto, 2025-04-17:
that's what this issue hoped to resolve, but nobody has implemented it & i don't have time to look now unfortunately.

kendallgoto, 2025-04-17:
in essence you just need to upload the bootloader and device factory provisioning / calibration data to the server while running the switchbota flow, then it can be restored later using a DFU package from the Switchbot website ...