kendallgoto/switchbota #42: SwitchBot with production code 2310 apparently bricked after flashing

christianhorn, 2023-05-13:
I have a SwitchBot model W2001400 for Japanese market, with production code 2221: that one could nicely be converted to Tasmota. Got another one this week with production code 2310: it automatically saw an available firmware update, fetched the prepared firmware (node index.js), and from then on went silent:

```
[root@kosmos server]# node index.js 
Server listening on port 80
::ffff:192.168.5.15 - /version/wocaotech/release.json
::ffff:192.168.5.15 - /version/wocaotech/release.json
::ffff:192.168.5.67 - /version/wocaotech/firmware/WoPlugJP/WoPlugJP_V1
::ffff:192.168.5.67 - /payload.bin
```

My best guess is that the internals have changed, and the Tasmota firmware did not get to run. Haven't opened the plug as of now. Just dropping this here, for potential ideas.

kendallgoto, 2023-05-13:
after restarting the plug, is it responsive at all as a regular plug (i.e. using the switchbot app)?

also it seems to be cut off in the log you sent -- maybe you have the full log -- the URL is `/version/wocaotech/firmware/WoPlugJP/WoPlugJP_V1` and missing the ending `x.bin`. Would be good to know which FW version it intended to pull

kendallgoto, 2023-05-13:
Also in your original comment, you described
> LED not blinking, can not be put into reset with pressing the button for 30seconds.

I would not expect either of these to be the case on a freshly flashed Tasmota device. Tasmota doesn't know what GPIO the button or LED are on when it is first flashed. So it will be fully unresponsive (it will look like a brick). You need to plug it in and connect to the tasmota configuration network to deploy the device-specific definition. Otherwise, it can't do anything except create a wifi network.

christianhorn, 2023-05-13:
> also it seems to be cut off in the log you sent -- maybe you have the full log -- the URL is `/version/wocaotech/firmware/WoPlugJP/WoPlugJP_V1` and missing the ending `x.bin`. Would be good to know which FW version it intended to pull
Right, I resized terminals here and that got cut off before I cut'n'pasted the output into a text file. If there is no further log with these requests, the information is probably lost.

Thanks a lot for the hint - at that point the plug is spanning up it's own AP, I had just looked for signs via Bluetooth, and in the wlan network I want it to use.
Sorry for the noise.
So 2310 also work, and we could extend the model line with the Japanese variant.

closed 2023-05-13 by christianhorn

kendallgoto, 2023-05-13:
No worries, thanks for testing. I'll look at adding the P/N to the compatibility list as well.