kendallgoto/switchbota #43: Just sharing my n00b experience (not an issue)

ams3401, 2023-05-31:
So I bought SwitchBot Smart Plug Mini 15A from Amazon and received them today (05/31/2023). They were about $7.00 each.

I spent nearly the entire day trying to get the DNS spoofing/redirecting/hijacking/whatever working because it just isn't a standard GUI-integrated function on my router (Asus RX-AX89X) or AP (Ubiquiti U6-Lite).

Trust me, I tried hot spotting my main PC to create an SSID and getting all devices including my cell phone connected, attempted using a Macbook, and tried all sorts of things just to get "www.wohand.com" ping at my own PC which is on standby to get the ping triggered.

What ended up being the solution was dividing a partition on my sub-pc and installing Ubuntu on it to get Pi-hole up and running.
On Pi-hole, I went on Local DNS -> DNS Records -> put "www.wohand.com" and "wohand.com" in the domain box and IP address of my PC running the server (node index.js). 

By the way, to get this server thing running on Windows, you need to install https://nodejs.org/en/download/ and run the command prompt. Open CMD, navigate to the server folder (e.g. cd C:\Users\user\Desktop\switchbota-main\server) and hit enter, then type node index.js and you'll see the thing waiting for its thing).

After I dealt with the DNS stuff, the rest went fairly smoothly.

Plug in the smart plugs into the outlet, download Switchbot App, pair them and don't update the firmware, get their BLE address in cog icon -> device info (looks something like this: 68:B6:B3:B3:D0:62). At this point, the server prompt window should say "::ffff:YOURIP - /version/wocaotech/release.json". If it still says "Server listening on port 80" and nothing else, gotta go back and do the DNS shit properly.

Then you need to download another app called nRF. On there, you need to click the Scanner tab and start scanning. After the provisioning is finished, you'll see multiple bluetooth devices from which you need to find the matching BLE address of your device. Once you locate it, hit connect, click "Unknown Services", two more stuff will unfold, click the upward arrow icon next to "Unknown Characteristic".

When a new window pops up, select Byte Array in the drop list, then right left of it, where there's a field to enter shit, hit it and type 57 0F 0A 01 0D (0 is zero not "O") and hit send. You'll see "::ffff:YOURIP - /version/wocaotech/firmware/WoPlugUS/WoPlugUS_V13.bin" on your server prompt window. 

Wait about a minute, on your phone again, press the same upward arrow icon again and this time, type 57 0F 0B and hit send and you'll hear the click sound from your plug. It's the sound of success. 

Now the window will say "somethingsomething /payload.bin" and wait till you see "tasmota-something" SSID in the list of Wifi to connect to on your PC. Click connect, go to 192.168.4.1 on your internet browser. From then, you'll do the usual configuration of Tasmota and it's done. 

The reason why I wrote this is because I came to this repo with 0 knowledge and it was extremely difficult to follow even with the provided guide, so I hope somebody else doesn't go through the same misery. Also, big thanks to DigiblurDIY. Without his video, I would've just returned these plugs and get Zigbee compatible ones. Here's the video guide I referenced: https://www.youtube.com/watch?v=iTexFQ0Th0I

jjcf89, 2023-09-27:
This helped a lot thanks

kubedzero, 2023-12-20:
Thanks for sharing these notes, I've collected them and formatted them into a documentation rewrite and have a PR sent for review https://github.com/kendallgoto/switchbota/pull/48

DOT850, 2024-01-21:
Yes! Thanks for the notes. Very helpful and helped to streamline the process.

AJolly, 2024-10-13:
One more note - you don't need to do anything with bluetooth.  Just need to set the correct DNS entries and you can update the fw in the app. You may need to click on the plug, then settings, then fw update. Just flashed 4 this way.

closed 2024-12-06 by kendallgoto