kendallgoto/switchbota #44: SwitchBot Strip Lights

donavanbecker, 2023-07-05:
Any luck with this working on the SwitchBot strip lights?

kendallgoto, 2024-07-17:
the strip lights were discontinued last year + I haven't personally used them at all, so I couldn't say for sure. However, they use an ESP32-C3 architecture similar to the plugs/bulb, so there's a good chance that it would work (but, they came out in 2021 before the plugs, so the system might have some differences / nuances)

however, this project is really geared towards avoiding disassembly of products that can't be safely reassembled, e.g. products that operate at 120VAC (bulb/plugs). The strip lights are DC operated and have a programming header built in, so it would be far easier to just open them up and program the flash for Tasmota rather than try to OTA.

reference: https://fccid.io/2AKXB-W1701100/Internal-Photos/Internal-Photos-5475063

donavanbecker, 2024-07-17:
Good to know! Thanks!

closed 2024-07-17 by donavanbecker