kendallgoto/switchbota #46: Trouble with current switchbot

maskedpirate, 2023-10-17:
I picked up a 4 pack of these guys. I've started with the first one. I have the node server running and I can curl www.wohand.com from another host on the same network and see that it makes a request to the node server (i.e. I feel like my dns resolver is configured correctly). I've then paired a switchbot with my phone. The app shows it's on v1.4 and that an update is available. However, I never see any traffic to the node server from the switchbot. I also connected to the switchbot with nrf and sent the first Bluetooth payload and still nothing on the node server.
I should also mention I did a packet capture from my pfsense firewall of all traffic from the switchbot, including reverse DNS resolution and I never saw it try to resolve wohand.com, nor did it try port 80 outbound to anywhere. It was talking to some AWS hosted servers, but not on 80.
Could anyone confirm recently purchased switchbots (W1901400) are still able to use this technique. Or provide some additional guidance on where to start?
Thanks

stncttr908, 2023-10-17:
The process in the readme worked perfectly for me a few months ago, and as far as I know there have not been any firmware updates since which would have broken it. For some reason it is not using your pfSense DNS. Do you see any other outbound DNS requests from the plug and/or Switchbot app? Are you sure you formed the Bluetooth packet properly?

maskedpirate, 2023-10-17:
I can't say with 100% that I formed the Bluetooth packets correctly, but thought I followed it.
I didn't watch packets from the switchbot app. I'm away from home until this evening and I'll check that out as well.
Tonight I'll also disable mobile data on my phone when working it to make sure DNS isn't escaping another path.
Thanks!

maskedpirate, 2023-10-18:
So good news; it was a PEBKAC issue. I unfortunately was "testing" connectivity to the node server from another VM on the same host which was sharing a network address. i.e. Traffic was flowing in a loopback and never ingress to the windows OS. So as soon as I realized this, I enabled port 80 ingress to my windows host and everything clicked into place.

I appreciate the feedback from the community!

I know it's pretty basic, and should be common knowledge, however it may help others to have a reminder in the readme to ensure their host firewall allows traffic inbound to the node server.

Thanks again!

closed 2023-10-18 by maskedpirate

stncttr908, 2023-10-18:
Happy to hear it worked. Enjoy!