kendallgoto/switchbota #50: Upgraded to v18 and bricked all 3 units

Draagynslair, 2024-06-15:
Followed the directions, verified connection to node.js server. Navigated to update firmware inside the Switchbot app. Everything went great, until the switches power cycled. Now nothing no power. Dead?
Updated from v15 (factory) to v18.

Node output:
Downloading missing binary https://github.com/arendst/Tasmota/releases/download/v11.1.0/tasmota32c3.factory.bin
Downloading missing binary https://github.com/kendallgoto/switchbota/releases/latest/download/app.bin
Server listening on port 80
::ffff:192.168.1.19 - /
::ffff:192.168.1.152 - /version/wocaotech/release.json
::ffff:192.168.1.152 - /version/wocaotech/release.json
::ffff:192.168.1.135 - /version/wocaotech/firmware/WoPlugUS/WoPlugUS_V18.bin
::ffff:192.168.1.135 - /payload.bin
::ffff:192.168.1.147 - /version/wocaotech/firmware/WoPlugUS/WoPlugUS_V18.bin
::ffff:192.168.1.147 - /payload.bin
::ffff:192.168.1.64 - /version/wocaotech/firmware/WoPlugUS/WoPlugUS_V18.bin
::ffff:192.168.1.64 - /payload.bin
::ffff:192.168.1.152 - /version/wocaotech/release.json

kendallgoto, 2024-06-15:
that looks fine to me. did you check for wifi networks? the plugs look "dead" after a successful upgrade. but they create a wifi network that you can connect to and configure them

Draagynslair, 2024-06-15:
I can't even get them to pass electricity through with the built in button.

kendallgoto, 2024-06-15:
yes, again, did you check for wifi networks? no functionality of the plug will work after you upgrade it except the wifi until it is configured with the hardware profile. tasmota has no idea how to interact with the hardware until you configure it

Draagynslair, 2024-06-15:
Yes, there IS a WIFI tasmota ssid on plug in.
A note in the directions would be helpful, If one exists that I missed then I apologize.
Thank you.

kendallgoto, 2024-06-15:
yeah, it's come up before. the "bricked" looking behavior should probably be further clarified. glad you got your plugs working!

closed 2024-06-26 by kendallgoto