kendallgoto/switchbota #51: looping on payload.bin

mogorman, 2024-07-16:
sometimes when I try to flash my switchbot lightbulbs i get stuck in an infinite loop. it never seems to recover. if i unplug the bulb it will be bricked. is there anything that can be done to prevent or solve this? I see no difference in bulbs that are successful vs ones that end up in this state.

```
::ffff:10.0.2.152 - /version/wocaotech/firmware/WoBulb/WoBulb_V12.bin

::ffff:10.0.2.152 - /payload.bin

::ffff:10.0.2.152 - /payload.bin

::ffff:10.0.2.152 - /payload.bin

::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin

::ffff:10.0.2.152 - /payload.bin

::ffff:10.0.2.152 - /payload.bin

::ffff:10.0.2.152 - /payload.bin

::ffff:10.0.2.152 - /payload.bin

::ffff:10.0.2.152 - /payload.bin

::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin

::ffff:10.0.2.152 - /payload.bin

::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin

::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin

::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin

::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin

::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin

::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin
::ffff:10.0.2.152 - /payload.bin
```

kendallgoto, 2024-07-17:
generally, this happens because there is a failsafe in the firmware where it will try to download the payload, verify its checksum, then apply it. If it can't validate its checksum, it tries to redownload it again. Since the code works in-memory, it clears internal storage before downloading the payload to have a place to write to, so there's no possible recovery method except to try again and hope the firmware downloads 'correctly' (in practice on the plugs, this isn't actually necessary, but it's the most trivial method to design, at the risk of greater vulnerability to power loss issues).

That said, in practice I've never actually seen it fail and get a corrupt download. It's possible if the local binary that you have is corrupt in the first place, but it is validated by the switchbota server when it downloads (https://github.com/kendallgoto/switchbota/blob/main/server/index.js#L123C1-L124C1) so that is unlikely. The only thing I could possibly imagine is a very poor WiFi connection that is interrupting the downloads and causing it to have to retry. Also, this is on a WoPlug / Switchbota Bulb which I don't have any personal experience with + isn't officially supported by this project. Maybe others can comment on any special nuances for that bulb, but I doubt it is much different than the plugs.

My guess would be that some bulbs are further away from your WiFi AP and that explains the inconsistent behavior. In my experience, most ESP32 designs I've used have pretty spotty WiFi performance, with some products having intermittent connections at 10-20 ft line-of-sight away from an AP. Can you describe your networking + physical setup more?

mogorman, 2024-07-17:
it worked, just took a few hours of trying.  i recommend if anyone gets in this state just let it sit it eventually will finish successfully

closed 2024-07-17 by mogorman