kendallgoto/switchbota #52: Not seeing DNS Requests from App

daelsc, 2024-07-17:
I successfully set up a PiHole DNS that is successfully redirecting wohand.com and www.wohand.com to my local NodeJS server.  It redirects correctly on my PC and my phone web browser and I see the connection in the NodeJS terminal.  Unfortunately, I don't see any activity from the SwitchBot App.  Is it possible they hard-coded DNS servers in the latest version of the SwitchBot App?  Has anyone successfully done this recently?

kendallgoto, 2024-07-17:
the app doesn't decide or use the dns servers at all. the plug connects to wifi and connects to the DNS servers on its own. do you have the plug connected to the network and did you trigger an ota on it?

The app only asks the server (not through wohand.com) whether or not an update exists. If it does, the device itself fetches the file (from wohand.com, via the DNS proxy) once told to do so over BLE. I wouldn't ever expect to see activity from the app reflect in the dns logs.

daelsc, 2024-07-17:
i haven't triggered an OTA update as I was worried it might actually update with a SwitchBot firmware (currently have v1.5 on the plugs, the app says v1.8 is available)

I got stuck on this step:

Test the Node server functionality by opening the SwitchBot app and then looking at the terminal where Node is running. A new line should have shown up similar to ::ffff:192.168.0.101 - /version/wocaotech/release.json. This indicates the server was accessed instead of the real wohand.com and that it successfully served a JSON file to the app. The IP 192.168.0.101 is the IP of the phone running the app.

kendallgoto, 2024-07-18:
did you start the ota?

if the dns works on your phone and your phone is on the same network as the plugs, then it should be fine. if you don't trigger the OTA, nothing is going to happen.

kendallgoto, 2024-07-18:
and worse case, you can retrigger another OTA manually anyway, if it doesn't grab the right file. but regardless nothing will happen until you start the upgrade most likely (the app sometimes pulls a version file but it isn't really guaranteed)

0allenc0, 2024-07-18:
Stuck with the same issue for a few days now - tried the below

Firmware: 1.5 out of the box
- Switchbot App Add Device - Doesn't seem to connect to the Node.js server (nothing in terminal)
- Clicking on "Firmware upgrade" in Switchbot App gets to 10% and then stops - nothing in the Node JS terminal
- Triggering via BLE command (570F0A010C) - nothing in Node JS terminal

A small piece of progress is in getting it to downgrade firmware to 1.2
- Add Device in Switchbot App, setup with normal WiFi
- nrfConnect app - Connect and send BLE command (570F0A010C), wait a few minutes 
- Send 570f0b command and hear click indicating restarting

Theory: Switchbot OTA is looking in a different location (no longer wohand.com?)

daelsc, 2024-07-18:
Ok, I just needed to push the firmware and it all worked as expected.  Thank you for the help!

closed 2024-07-18 by daelsc

kendallgoto, 2024-07-19:
thanks for the update; sorry for the confusion on the write up. we'll have to get that cleared up

nwithan8, 2025-12-16:
>     * Clicking on "Firmware upgrade" in Switchbot App gets to 10% and then stops - nothing in the Node JS terminal

For future travelers, for me, I was connecting the Switchbot devices to a different WiFi network (different subnet/VLAN) that I forgot I had configured to not allow access to the subnet/VLAN that the computer running the NodeJS server was on. Temporarily instead connected the Switchbot devices to a network that did allow cross-VLAN communication, and the upgrade worked.

So, if your firmware upgrade process gets stuck at 10%, it's a good sign that means the device cannot communicate with the NodeJS server.