kendallgoto/switchbota #55: bricked ?

tismofied, 2024-08-21:
tried flashing tasmota on of the newer batches 2414 using the upgradde button on the app, I got this
```
Server listening on port 80
::ffff:192.168.20.82 - /version/wocaotech/release.json

::ffff:192.168.20.83 - /version/wocaotech/firmware/WoPlugJP/WoPlugJP_V20.bin

::ffff:192.168.20.83 - /payload.bin

::ffff:192.168.20.83 - /payload.bin

::ffff:192.168.20.83 - /payload.bin

::ffff:192.168.20.83 - /payload.bin

::ffff:192.168.20.83 - /payload.bin
```

in hindsite, I should used the bluetooth trick to upload the file. 
I can't get the plug to respond and there is no tasmota APs under wifi networks. ugh lol

kendallgoto, 2024-08-21:
others have had luck moving their plugs closer to their network AP etc so the network connection is more stable, since it will retry to download over and over again if the connection is poor. most esp32 designs have pretty terrible wifi antennas from my experience, coupled with a bad network stack. try that perhaps?

kendallgoto, 2024-08-21:
see #51 for example

tismofied, 2024-08-21:
> others have had luck moving their plugs closer to their network AP etc so the network connection is more stable, since it will retry to download over and over again if the connection is poor. most [...]

This is exactly what the problem was. Weak connection to wifi. I got impatient and must have either pushed the power button or held it in while it was looping the payload.bin lesson learned to never do that again lol. 
PS: so if I get no power on the plug when I plug things in it and pushing the power button doesn't do anything it means the plug is trash now right? 
I opened it and tried the soldered method with the serial flash but couldn't get it to connect.

kendallgoto, 2024-08-21:
is it producing a config network? the plug won't respond to the button & the relay will be left open until the device is configured via the webpage. so it may look bricked but be fine, just waiting for configuration

in any case, youll still always be able to use the hardwired flashing method.

tismofied, 2024-08-21:
> is it producing a config network? the plug won't respond to the button & the relay will be left open until the device is configured via the webpage. so it may look bricked but be fine, just waiting for configuration
> 
> in any case, youll still always be able to use the hardwired flashing method. 

There was no tasmota ssid showing. I left it plugged overnight and there was nothing.

kendallgoto, 2024-08-21:
not sure then; if if only happened to some of your plugs then I suppose it must have broken during the install and is bricked. i haven't actually ever had that happen on my 50 units, but it's certainly possible.

best of luck in flashing it; it should be recoverable. there isn't really a way to permanently blow up the chip during the firmware upgrade.

tismofied, 2024-08-21:
> not sure then; if if only happened to some of your plugs then I suppose it must have broken during the install and is bricked. i haven't actually ever had that happen on my 50 units, but it's [...]

It's no big deal really. Thank you for this great project.

closed 2024-08-26 by kendallgoto