kendallgoto/switchbota #56: DNS redirected but not showing up in node server

infinitenewworld, 2024-12-01:
I"ve redirected wohand.com to my node server but am not getting any line when opening up Switchbot. I'm wondering if perhaps the firmware server is now different. Is there anyway to check? Any clues would be so helpful!

kendallgoto, 2024-12-01:
the device calls wohand, not the app (though the app does / did in certain circumstances, but the device calling wohand.com is what matters)

infinitenewworld, 2024-12-02:
it seems my dns redirect is working as when I ping wohand.com and www.wohand.com I get my specified ip

kendallgoto, 2024-12-03:
did you start an update on the plug and not get the expected result? it won't call the url until you start an update

infinitenewworld, 2024-12-05:
that worked! thank you so much :)

infinitenewworld, 2024-12-05:
One last request, could you send me a copy of the partition wizard that worked with 11.1.0. It seems that the new version is not compatible with the older 11.1.0 so I can't fix the partition. Thanks!

infinitenewworld, 2024-12-05:
What I'm actually trying to do is to set this device up to work with ESPHome as a presence sensor using BLE. Not sure if this will be possible but hoping you may have some tips!

kendallgoto, 2024-12-06:
i don't actually have any personal experience w/ the partition wizard or esphome; i believe @kubedzero wrote that on the readme, maybe they can chime in here

kendallgoto, 2024-12-06:
there might also be some relevant changes in the documentation that you're looking for in #53

kubedzero, 2024-12-18:
> send me a copy of the partition wizard that worked with 11.1.0. It seems that the new version is not compatible with the older 11.1.0 so I can't fix the partition

It's been a little while since I did this last. I believe I had to update from Tasmota 11.1.0 to 12.5.0. I then downloaded the Partition Wizard application (either the latest version or the older versions if you encounter bugs or the application doesn't load). Those can be found at https://github.com/arendst/Tasmota/commits/development/tasmota/berry/modules/Partition_Wizard.tapp

Once the partition wizard runs and reconfigures, you should be ready to go to update to the latest version of Tasmota.