kendallgoto/switchbota #58: Compatibility with "new" switchbot plugs?

AJolly, 2025-05-22:
Any idea if the "2025 New" switchbot plugs that are now listed on amazon are compatible with this project?

kendallgoto, 2025-05-22:
I bought one and will take a look

ryanalden, 2025-05-26:
Yes, I can confirm that the process still works fine with the “2025” units (which ship with FW v1.8 & out of the box have a waiting update for v2.0)

jspiros, 2025-05-27:
Same here, bought a 4-pack from Amazon yesterday before @ryanalden updated, received them today and just flashed them with no issue. Showed up as v1.8, wanting an update to v2.0, quick and easy.

AJolly, 2025-06-05:
Btw, have any of you run into issues with stability with these plugs when using esphome? I'll notice that they will reboot themselves frequently.  Also I've had a few of them have their relays burn out. (But these still seem like the cheapest option for esphome smart plugs)